tagyoureit/nodejs-poolController #824: [BUG] Circuits with schedules that overlap in a polling window

CodeSlinger69, 2023-07-11:
### nodejs-poolController Version/commit

head

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

Restating one of my messed up pull requests as an issue, for tracking purposes.

This is not urgent, as the workaround is simple - allow for at least one minute between different schedule items that work on the same circuit.

[Here's the documentation](https://github.com/tagyoureit/nodejs-poolController/pull/813). This issue is still valid as of today.

### What happens?

[image]

### What should have happened?

Schedule & circuit should activate.

### Additional information

Changing end time to 23:59 will workaround this limitation.

rstrouse, 2023-07-11:
Are you sure you have updated your software.  I just tried this and got the following.
[image]

rstrouse, 2023-07-11:
And when I followed directions better I got this.

[image]

CodeSlinger69, 2023-07-11:
Yup, software is up to date. Just did a git pull, npm i, npm run build again to be sure, but same behavior.

I notice in your images above that you are testing at 12:42. The issue only occurs when two schedules try to change the same circuit in the same polling window. If you start up njsPC when the current time is in the middle of one of the schedules, all is good. Only when you watch njsPC transition across the adjoining time (in this case midnight), do you see the problem. This is hard to explain, but was harder to track down.

To see this happen:
1. Stop njsPC
2. Set your host OS clock to 23:59:00
3. Start njsPC

The 18:00-24:00 schedule will turn on upon njsPC startup, as expected.

In the polling window that occurs between 00:00:00 and 00:00:02, the 18:00-24:00 schedule will turn off (as expected), and the 00:00-15:00 schedule will turn on (as expected).

In the next polling window between :03 and :05, the 00:00-15:00 schedule will turn off (NOT expected). 

At this point, the pool circuit is off, even though the schedule says it should be on. 

BTW, I fixed this for myself by adding this line to controller/nixie/schedules/Schedule.ts, line 214, but you mentioned that in another thread that this will probably not the right way to solve for this. And as you commented in the code :)
```
            else if (shouldBeOn && this.running) {
                // Check to see if circuit is on, if not turn it on.
                // RKS: 07-09-23 - This was in PR#819 buut this needs further review since the circuit states are not to be set here. This would
                // trash delays and manualPriority.
                // if(!cstate.isOn) ctx.setCircuit(circuit.id, true);
```

rstrouse, 2023-07-11:
That makes much more sense now.  I'll check it out.  All of the schedules for the given circuit should be evaluated and only turned off after the evaluation.  This is probably where it is falling down.

CodeSlinger69, 2023-07-16:
Nice work on this one too! I've confirmed that the issue as I described in this thread is corrected. I'll close it. Thank you!

closed 2023-07-16 by CodeSlinger69