tagyoureit/nodejs-poolController #858: Flow sensor from Pentair VSF

trmb-vast, 2023-10-14:
## 💥 Proposal

### What feature you'd like to see

for ORP and PH pump, I'd like to use a flow(rate) sensor already available from the pentair-VSF pump via vixie.
## Motivation

I have a ORP pump already, Now that I'm adding a PH pump, it's now more important to dose only when the pump is flowing, but I don't want to add another sensor when I've got data from the VSF pump itself.
## Pitch

Flow sensors are important for dosing pumps. but I want to use the flow data from my VSF pump.  eg: "only dose when flow above 15 gpm"..   existing feature seems to only support external REM switches and flow meters.

jeffegg, 2023-10-14:
I have changes for REM to support 4-20ma flow rate sensors (generic device like a temp sensor), that I need to put in a Pull Request(PR) to get in the mainline. I wanted to add this to my nixie dash panel as well just haven't had a time to look at the changes needed for on the njsPC side. So, I second the ability to have flow rates there.

tagyoureit, 2024-06-10:
Closing due to lack of activity.  If you are still working on this, please reopen.

closed 2024-06-10 by tagyoureit