tagyoureit/nodejs-poolController #862: [BUG] Can't turn off heatmode

Tegguy, 2023-11-06:
### nodejs-poolController Version/commit

8.0.1

### nodejs-poolController-dashPanel Version/commit

8.0.0

### Node Version

V16.19.1

### Platform

Linux pool-controller 6.1.19-v7+ #1637 SMP Tue Mar 14 11:04:52 GMT 2023 armv7l GNU/Linux

### Are you using Docker?

- [ ] Yes.

### OCP

Intellitouch 8

### Pump(s)

2X Intelliflo

### Chlorinator(s)

IC-40

### What steps will reproduce the bug?

When trying to change heatmode to "OFF" it doesn't switch on the dashpanel. This happens with both Pool and Spa

### What happens?

Not change in heatmode

### What should have happened?

Heatmode should switch to "off"

rstrouse, 2023-11-06:
Please perform a replay capture of changing the heat mode.  I need to see the responses from the panel.

Tegguy, 2023-11-06:
Attached is the packet Capture (I hope I did it correctly)
[Bug 862 Replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/13272621/Bug.862.Replay.zip)

rstrouse, 2023-11-06:
Restart njsPC and try the capture again.

Tegguy, 2023-11-07:
rebooted njsPC and restarted the dashpanel and it appears to be working. Sorry for opening the bug.

closed 2023-11-07 by Tegguy