tagyoureit/nodejs-poolController #869: [BUG] Lights Stuck

matthewusf, 2023-11-27:
### nodejs-poolController Version/commit

8.0.1

### nodejs-poolController-dashPanel Version/commit

8.0.0

### Are you using Docker?

- [ ] Yes.

### OCP

Intellicenter i5PS - 1.064

### What steps will reproduce the bug?

Selecting a light color produces stuck in "Sequencing Theme/Color Operation... Please Wait!" (Intellicenter -> Microbrite)

### What happens?

[image]

### What should have happened?

I successfully changed the light within intellicenter, restarted njsPC and it is still stuck in this mode.

### Additional information

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/13479079/replay.zip)

dsisk112, 2023-12-30:
I am having the same issue. I do get an error trying to modify the color. Circuit or Feature Id not valid. 
[image]

It did this with 7.x as well. Now running 8.0 UI and 8.0.1 backend.

dsisk112, 2023-12-30:
I do have two null circuits in the Light Group. The third circuit is the real Light circuit. I assume I could delete these at the pool controller. Or they are not really there. I will check. 

[image]

rstrouse, 2024-01-16:
@dsisk112 this is part of the goofiness in Touch firmware that it will sometimes put invalid circuits in the IntelliBrite group.  I assume you have only 1 light so try deleting the two invalid ones and save the group.

@matthewusf Ill have a look at the messages from IntelliCenter.  This flag normally comes from IntelliCenter when it is attempting to sequence one of its groups.  Perhaps there is a difference from your panel.

rstrouse, 2024-01-16:
@matthewusf also please change the theme on the IntelliCenter panel.   Let’s see if it clears the flag.

rstrouse, 2024-01-16:
@matthewusf can you capture a scenario where you are turning a circuit on and off.  I do not see any returns from IntelliCenter when it asks for the circuit status in the log output but there were no messages after the request.  I assume this is because you stopped the capture before IC responded.

matthewusf, 2024-01-25:
Just got back in town, really appreciate your help. Apologies if I wasn't very clear but I seem to be having a different issue than the person commenting on this bug. My issue is this: 

A couple of months ago, I changed the light color from within njsPC and got "Sequencing Theme/Color Operation... Please Wait!". I do not remember if I got any errors after that, but it did not change the light color. I had to change it within Intellicenter. Ever since, njsPC is stuck in that same "Sequencing" state. If I change the color in Intellicenter, it correctly recognizes the color change by changing the light bar to the proper color, but the color pallet continues spinning and says "Sequencing" if you click on it (preventing me from selecting a color within njsPC).

My guess is the capture isn't capturing anything since it's not allowing me to even select a color change, if that makes sense. My thought is there's gotta be a way to have it just rebuild its configuration - but when I select "Rebuild Configuration," it just gets stuck at 34% and never actually loads. I have to then restart njsPC to get it to show "Ready" again. It's definitely on the latest versions for njsPC and dashPanel.

rstrouse, 2024-01-26:
I don't want you to select a color I want you to turn the group on and off or at least another circuit.  The original output did not include the state response as the last message was the request.

matthewusf, 2024-01-29:
Appreciate it - does this give you what you need to see it? I turned the lights on and off a couple of times:

[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/14087054/replay.zip)

rstrouse, 2024-01-30:
Thanks that was really helpful.  Pull njsPC it appears IntelliCenter does not report when it finishes hold or recall if the circuit goes off before it finishes.  The software will now initialize these on startup and clear the sync when the circuit is off.

rstrouse, 2024-01-30:
Create a caption with it reloading the config from the start.  It is waiting for the panel to respond with some information.

matthewusf, 2024-01-30:
Interesting, here it is after being stuck at 15% for a good hour:

[replay (5).zip](https://github.com/tagyoureit/nodejs-poolController/files/14105315/replay.5.zip)

tagyoureit, 2024-06-10:
Closing due to lack of activity.  If this persists, please reopen.

closed 2024-06-10 by tagyoureit