tagyoureit/nodejs-poolController #881: [BUG] Intellibrite Light Group with Relays

truvec, 2024-01-11:
### nodejs-poolController Version/commit

8.01 0501e77

### nodejs-poolController-dashPanel Version/commit

8.00 e3ff402

### relayEquipmentManager Version/commit

?? Latest

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

Setup Relay board. Add aux-Circuits as intellibrite. Relays function as normal individually. Create a Light group. When controlled via group the relays do not turn off or on properly. If a circuit group is created the relays function as a group. Light group sync functions work just not the on/ off toggle.  
[image]
[image]
[replay-2.zip](https://github.com/tagyoureit/nodejs-poolController/files/13909652/replay-2.zip)

### What happens?

Relays do not toggle on/off as a group.

### What should have happened?

Shoould turn all on/off

truvec, 2024-04-04:
the other issue is that setting intellibrite colors either individually or as a group does not create the desired color. I have changed the delay in circuit.ts to 500 and that seems to get the individual lights to set as prescribed. Group settings do not fall in line.  

Here is a replay. 

[Archive.zip](https://github.com/tagyoureit/nodejs-poolController/files/14860063/Archive.zip)

`protected async setIntelliBriteThemeAsync(cstate: CircuitState, theme: any): Promise<InterfaceServerResponse> {
        let arr = [];
        if (cstate.isOn) arr.push({ isOn: false, timeout: 500 });
        let count = typeof theme !== 'undefined' && theme.sequence ? theme.sequence : 0;
        if (cstate.isOn) arr.push({ isOn: false, timeout: 500 });
        for (let i = 0; i < count; i++) {
            if (i < count - 1) {
                arr.push({ isOn: true, timeout: 100 });
                arr.push({ isOn: false, timeout: 100 });
            }
            else arr.push({ isOn: true, timeout: 500 });`

tagyoureit, 2024-04-04:
Can you run a couple tests first.  Set the timeout back to 100ms and turn off all the lights.

Try to turn on one individual light to a color/theme and see if it works.  If it doesn't, turn all the lights off and try again with the 500ms delay.

Again turn all the lights off.  Try setting a light theme now.  I'm wondering if there is a timing issue between an existing light being on and trying to set a theme (which tries to turn each light off first to synchronize them).

tagyoureit, 2024-04-04:
Actually, just spent some time going through the code and made some adjustments to the timing.  Can you pull the changes I made and see if that helps.

tagyoureit, 2024-04-16:
Checking to see if you had a chance to test this out.

tagyoureit, 2024-05-01:
Closing this.  Please re-open if it's not fixed.

closed 2024-05-01 by tagyoureit