tagyoureit/nodejs-poolController #926: Measuring water level

katodude, 2024-04-17:
I would like to be able to measure water level, or tank level of a dispenser. These e-tapes are pretty popular and provided a resistance level that the REM can read from MegaBas card. 

https://www.adafruit.com/product/464?gad_source=1&gclid=CjwKCAjww_iwBhApEiwAuG6ccF35EknNk3dofOXtgYtgGOM3WIQHqU-V2aPSiSszRJ6AaCpusIqaXxoCOdQQAvD_BwE

Based on the level I can get a warning or turn on an autofill. 
-->

johnny2678, 2024-04-17:
I've been thinking about this too, but for my pool, so I can simulate an autofiller, which i don't have.  

Only reservation about using this for chem tanks is you don't know how long it will last in acid/bleach.

katodude, 2024-04-17:
Yep, I get it on the chem tanks. 

Another use case, and the primary reason for this is for a spillover in an infinity pool. Measuring the spillover basin to decide how fast to run the pump to pump water back into the pool.

I would also like to use it in my pool to measure evaporation. It will help with leak detection. Also when I am traveling I can use it to help with an autofill which I dont have either. I can hook up a hose with a smart valve to cycle on and off.

johnny2678, 2024-04-17:
exactly this.  Love monitoring the evaporation/overnight water loss idea.

Have you thought about how you would install it?  I was thinking about mounting it in my skimmer, and maybe enclosing in a 1.5 inch PVC to minimize splashing/swells when we're in the pool?  Not sure how to actually attach it - hot glue or liquid nails? 

I guess another issue is how to wire it.  My pad is about 30 feet from my pool so hardwiring directly to REM is out.  Had thought about maybe a battery powered PicoW device waking up every 5 minutes for a reading? Didn't really get much further than that when thinking about it.

tagyoureit, 2024-04-17:
There have been lots of conversations over the years about e-Tape and ways to measure chemicals or fill the pool.  Most of the products out there are low quality, low resolution, or inconvenient to install.  Here's [one](https://github.com/tagyoureit/nodejs-poolController/discussions/542).  Maybe the new e-Tape is better, but at 12" it would be hard to measure the level in any sizeable tank.

katodude, 2024-04-18:
At this point discard the tank idea. It’s more about measuring water depth. I have no idea if the products have improved. 

Wiring it will be a challenge but not insurmountable. You could potentially use the pool light conduit.

jwtaylor310, 2024-06-15:
I use a YoLink LoRa Water Level Monitor (Model YS7904F-UC) to alert me when the water level in my pool is low.  It is a simple float mechanism that has two states - "OK" and "Low".  I have mounted the transmitter module in a standard waterproof outlet box and the float is in a "T" shaped PVC pipe attached to the box.   The assembly sits on the edge of the pool with a bit of Velcro to keep it in place.  The device operates on two AAA batteries which are good for 2-3 years.  It transmits to a hub using LoRa (Long Range radio) with a range of up to 1/4 mile.  The hub requires an Internet connection and standard 120VAC power and can be located anywhere inside a nearby building.  It integrates easily with Alexa and Home Assistant.  It has a documented API which would allow it to added to the NodeJS program but the API interface is non-trivial and requires a User Access Credential unique to each user (API at http://doc.yosmart.com/docs/overall/intro).  A package of the Water Level Monitor and a Hub hub  costs $59.99 on Amazon.  YoLink makes lots of other devices for home automation and they all share a single hub.  I have found the YoLink products to be very versatile and reliable.  My pool level sensor has been in use for over a year without any problems.  I use the YoLink app on my phone to send me an alert if the water level is low.  Here's a photo of my installation:

[image]

Be advised that if you search for the sensor by part number, Amazon goes to the wrong product -- a YoLink leak sensor instead of the level sensor.  Search for "YoLink Water Level Sensor" in Amazon instead of the part number.

thrystan, 2024-07-17:
I tried the etape for water level measurement and I couldn't get it to give me reliable measurements in a 1" pvc pipe connected to my old Polaris port. I'm trying the [VL53L4CX](https://www.adafruit.com/product/5425) time of flight laser sensor now, though I don't have it working yet either. 
For sensing the level on my 15 gallon chemical tanks I am using these. https://www.amazon.com/gp/product/B07Z64CSLQ non contact level sensors. 
[image] I fasten them to the outside of the tanks with vhb tape and they give me 3+, 6+, 9+, and 12+ gallon readings for chlorine, which is granular enough for me. On my acid tank, which I dilute 4:1, so 3 gallons of acid in a full 15 gallon tank, I only have 2 sensors 5+ and 10+ gallons. I've been using the same tank sensors for 3 years now without any problems.