tagyoureit/nodejs-poolController #927: Allow missing chem sensors / schedule super-chlorination

jamestalmage, 2024-04-17:
I did a soft open here in Michigan. Water temp is 58˚, and saw some algae getting started on the shallowest step, so definitely was the right time to get some chemicals in, but there are plenty of opportunities for a freeze for almost the next month. Nightly temps will be 35˚ all next week. Not worried about my plumbing or chlorinator, as I'll just be using freeze protection settings in njsPC to keep the water moving... But my understanding is that it's probably dangerous to install my industrial combo PH/Orp/Temp sensor this early when there's a potential for it to freeze. I'll be circulating water through the flow cell, but my understanding is the body of the sensor (which is out of the water) would still be in danger from a freeze.

I'd like to just tell the system to "run at 100% chlorine production for X hours per day"... Measure with my old school drop reagents. Yes, more work, but with a light blocking cover on, and no bather load, not a ton of effort. Then in a few weeks, once I'm past the freeze danger, calibrate and install my sensor.

I see I can click "super-chlor" in dashpanel, and it does what i need. But can't seem to schedule it.

tagyoureit, 2024-04-17:
Correct, you can't schedule superchlor but that only sets the chlorinator to 100%.  Superchlor actually overrides the pool circuit and doesn't allow it to be turned off.  Hence, you can't schedule superChlor either.

Easiest thing to do is just disassociate the chlorinator from set the setpoints manually.  The % is a percentage of time so if you want the chlorinator to be on for 4 hours and your pool runs for 8 hours a day adjust the setpoint to 50%.  

[image]

jamestalmage, 2024-04-17:
I disabled ORP... as you showed...

With the setpoint set to 100%, it still isn't coming on.

[image]

(Ignore the filter pressure. I think the transducer is broken. Doesn't match the guage. Filter does need a cleaning after opening, but it's not that bad)

tagyoureit, 2024-04-18:
It could be a couple of things.

1. You pump shows 2207 rpm but 0 gpm.  Is it a VSF?  Why is it not registering flow?
2. Do you have super chlorinate enabled?  You can't change the set point when it is super chlorinating.

If you want to upload a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution#-dashpanel) (do not select reset configuration) I can see what the logs and your configuration tell me.

jamestalmage, 2024-04-18:
0gpm was me turning it off and back on and it wasn't primed yet... Once it primes it registers. Option to change setpoint is still grayed out.

jamestalmage, 2024-04-18:
[replay.zip](https://github.com/tagyoureit/nodejs-poolController/files/15019370/replay.zip)

tagyoureit, 2024-04-18:
In your './data'poolConfig.json` the chlorinator is disabled.  I'd have to dig in to see how this would even get set, but you can either delete the chlorinator and re-add it, or stop the app and manual change this to false.

```JSON
  "chlorinators": [
    {
      "id": 1,
      "disabled": true,  // should be false
```

jamestalmage, 2024-04-18:
That fixed it. 

Would be nice to have it setup so I didn't have to blow away my ORP settings to get the chlorinator to function without the ORP sensor present. 

How are folks in cold weather climates handling this scenario? Just opening later? It's my first time opening this early and I'm never going back. Even if it causes some difficulties with njsPC, opening to a clear pool is so much easier than fighting back an established algae bloom.

tagyoureit, 2024-04-18:
> Would be nice to have it setup so I didn't have to blow away my ORP settings to get the chlorinator to function without the ORP sensor present.

https://github.com/tagyoureit/nodejs-poolController/issues/927#issuecomment-2062411786

closed 2024-04-18 by tagyoureit