tagyoureit/nodejs-poolController #961: [BUG] Pump stays on after Pi shutdown

celestinjr, 2024-06-03:
### nodejs-poolController Version/commit

8.0.1/6d1ddbe

### nodejs-poolController-dashPanel Version/commit

8.0.0/e1e5190

### Node Version

v16.16.0

### Platform

Linux nixie-poolcontroller 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

### Are you using Docker?

- [ ] Yes.

### OCP

Nixie / Cannibalized Aqualogic AQL-PS-8

### Pump(s)

Pentair Challenger 1 Speed

### Chlorinator(s)

AquaRite

### What steps will reproduce the bug?

Set up two Intellibrite lights using two separate circuits.  Create a light group combining the two Intellibrite light circuits.  Attempt to set the color scheme on the light group.

### What happens?

The sequence of on/off signals occurs for each light, one light at a time, but the correct theme is not set.  Setting the color scheme via either of the individual light circuits works as expected.

### What should have happened?

All lights in the light group should be set to the chosen color scheme.

### Additional information

Just a guess, but it appears the color setting for the light group leaves the light off after the on/off sequences.  Perhaps it needs to leave the light on for a minimum duration for the new color scheme to take affect.

tagyoureit, 2024-06-09:
There were some changes made to light group timings recently.  https://github.com/tagyoureit/nodejs-poolController/commit/258b45c780f9df901daf5ded6a9cdaaa723940a4

Have you pulled the latest

celestinjr, 2024-06-10:
Yes, as of original posting, I was on 8c079e5ae8602694d459774966eaaba926d1ddbe, which appears to be (2) commits newer than 258b45c780f9df901daf5ded6a9cdaaa723940a4.

celestinjr, 2024-06-15:
Here it is. Colors were set correctly when set individually. Neither light changed color after attempting to set the theme via the light group.

[2024-06-14_16-06-55.zip](https://github.com/user-attachments/files/15845478/2024-06-14_16-06-55.zip)

tagyoureit, 2024-07-01:
Hi, just getting back to this.  I'm looking at your logs...

```
// single light change
[14/06/2024, 20:38:50] info: [20:38:50] 192.168.10.49 PUT /state/circuit/setTheme {"id":2,"theme":4}
[14/06/2024, 20:38:50] info: Setting light theme for Pool Light to red [10]

// single light change
[14/06/2024, 20:38:58] info: [20:38:58] 192.168.10.49 PUT /state/circuit/setTheme {"id":3,"theme":4}
[14/06/2024, 20:38:58] info: Setting light theme for Spa Light to red [10]

// group light change
[14/06/2024, 20:39:09] info: [20:39:09] 192.168.10.49 PUT /state/circuit/setTheme {"id":193,"theme":2}
[14/06/2024, 20:39:09] info: Setting light theme for Pool Light to blue [8]
[14/06/2024, 20:39:09] verbose: REM server request initiated. PUT /state/device/gpio:1:11 [{"isOn":false,"timeout":1000},{"isOn":false,"timeout":1000},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":1000}]
[14/06/2024, 20:39:14] info: Setting light theme for Spa Light to blue [8]
[14/06/2024, 20:39:14] verbose: REM server request initiated. PUT /state/device/gpio:1:12 [{"isOn":false,"timeout":1000},{"isOn":false,"timeout":1000},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":100},{"isOn":false,"timeout":100},{"isOn":true,"timeout":1000}]
```

Are you not seeing the colors change (flash on/off) at all when you change the group?  Or are they just not changing to the correct colors?

celestinjr, 2024-07-01:
The lights never flash while cycling power, only once power remains on for a second or two.  I believe that's expected operation, at least for the 120V version.  

I have confirmed via the GPIO output that the power is cycling on/off using both methods.  My guess is that when setting the theme via the light group, the power is being cut and left off after setting each light, which is not allowing the theme selection to stick.  I'm assuming this was a result of attempting to keep the lights in sync.

Could/should we leave each light on for a couple seconds after setting the theme and before cycling the power together for a few seconds to sync?

tagyoureit, 2024-07-01:
You can play with the code a bit... I haven't heard other people having this issue, though, so wondering if there's something different here.

This is the code that runs the group settings.
We set each light via it's own sequence, and then turn it off, wait 5s (until all lights are off), and then turn them back on.  This should in theory sync them.  See my comment in the code below for where you could put a pause in before turning the lights off.

https://github.com/tagyoureit/nodejs-poolController/blob/master/controller/boards/NixieBoard.ts#L1307
```js
public async setLightGroupThemeAsync(id: number, theme: number): Promise<ICircuitState> {
        const grp = sys.lightGroups.getItemById(id);
        const sgrp = state.lightGroups.getItemById(id);
        //grp.lightingTheme = sgrp.lightingTheme = theme;
        let thm = sys.board.valueMaps.lightThemes.transform(theme);
        sgrp.action = sys.board.valueMaps.circuitActions.getValue('lighttheme');

        try {
            // Go through and set the theme for all lights in the group.
            for (let i = 0; i < grp.circuits.length; i++) {
                let c = grp.circuits.getItemByIndex(i);
                await sys.board.circuits.setLightThemeAsync(c.circuit, theme);
                // you could put `await setTimeout(3000);` here for a pause for 3s before turning off the light.
                await sys.board.circuits.setCircuitStateAsync(c.circuit, false);
            }
            await setTimeout(5000);
            // Turn the circuits all back on again.
            for (let i = 0; i < grp.circuits.length; i++) {
                let c = grp.circuits.getItemByIndex(i);
                await sys.board.circuits.setCircuitStateAsync(c.circuit, true);
            }
            sgrp.lightingTheme = theme;
            return sgrp;
        }
        catch (err) { return Promise.reject(err); }
        finally {
            sgrp.action = 0;
            sgrp.emitEquipmentChange();
        }
    }

closed 2024-07-02 by tagyoureit