tagyoureit/nodejs-poolController #971: Sensor Calibration Settings Don't Change

jwtaylor310, 2024-06-12:
### nodejs-poolController Version/commit

8.0.0 a80df29

### nodejs-poolController-dashPanel Version/commit

8.0.0 cb05840

### relayEquipmentManager Version/commit

N/A

### Node Version

v19.8.1

### Platform

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### RS485 Adapter

Elfin

### Are you using Docker?

- [ ] Yes.

### OCP

EasyTouch2 8

### Pump(s)

Intelliflo VSF

### Chlorinator(s)

Intellichlor-40

### What steps will reproduce the bug?

Open Settings ... General ... Sensor Calibration.  Change "Adjusted Value" of Water Sensor to correct value (test case - changed 80 to 88).  Click on "Save Calibration".  Adjusted Value reverts to previous value (e.g., 80).  Attempts to change Air Sensor calibration fail in the same way.  

### What happens?

Attempts to change "Adjusted Value" of Water or Air Sensor fail with value reverting to original immediately after clicking on the "Save Calibration" button.

### What should have happened?

"Adjusted Value" should change to and remain at the newly entered value.

### Additional information

Packet Capture: 
[replay.zip](https://github.com/user-attachments/files/15796050/replay.zip)

jwtaylor310, 2024-06-12:
Packet Capture is for attempted change of Air Sensor +10 degrees, no change to Water Sensor

tagyoureit, 2024-07-01:
It doesn't look like we were able to figure this on out... the 168 packet (I believe) is supposed to set this, but on my previous EasyTouch system I wasn't ever successful with it.

```
[6/11/2024, 7:51:34 PM] info: [7:51:34 PM] 192.168.80.1 PUT /config/general {"options":{"airTempAdj":10,"waterTempAdj1":0}}
[6/11/2024, 7:51:34 PM] debug: Assigning message id 242
[6/11/2024, 7:51:34 PM] info: {"port":0,"id":242,"valid":true,"dir":"out","proto":"broadcast","pkt":[[],[255,0,255], [165,30,16,33,168,9], [0,0,0,254,0,0,0,0,0],[2,163]],"ts":"2024-06-11T19:51:34.907-0400"}
[6/11/2024, 7:51:34 PM] verbose: Wrote packet [Port 0 id: 242] [255,0,255,165,30,16,33,168,9,0,0,0,254,0,0,0,0,0,2,163].Retries remaining: 3 
[6/11/2024, 7:51:35 PM] debug: Assigning message id 243
[6/11/2024, 7:51:35 PM] silly: Message in 243 is a response for message out 242
[6/11/2024, 7:51:35 PM] info: {"port":0,"id":243,"valid":true,"dir":"in","proto":"broadcast","for":[242],"pkt":[[255,255,255,255,255,255,255],[255,0,255],[165,30,33,16,1,1],[168],[1,158]],"ts": "2024-06-11T19:51:35.013-0400"}
```

I tried setting this from the OCP and seeing if it would broadcast the calibration values, but it never did.  And further, the instructions say the system must be in Service Mode (for whatever silly reason) and then contradicts itself by telling you that you only need to set one sensor but the instructions say to set the calibration for both pool/spa.

I'd change this on your OCP and see if you can first even be successful there.  If you have an ICP, then the next step would be to do a packet capture see if you can replicate the settings.  Otherwise, I don't have any way to test this currently because I no longer have access to an EasyTouch system.

jwtaylor310, 2024-07-02:
Thanks for the follow-up.  Today I tried adjusting the calibration settings on the OCP and the settings evert after entering them, so it looks like the problem is in the controller, not your software.  I did a Google search and this appears to be a common problem with no apparent solution.  My water temperature has been displaying 12 degrees hotter than reality.  I replaced the sensor this afternoon and the replacement reads within a degree of my reference thermometer, so that fixes my immediate problem.   My apologies for wasting your time on this!

closed 2024-07-07 by tagyoureit