tagyoureit/nodejs-poolController #972: [BUG] Schedule not turning on after first day

celestinjr, 2024-06-13:
### nodejs-poolController Version/commit

8.0.1/0e291ca

### nodejs-poolController-dashPanel Version/commit

8.0.0/d4f24a2

### relayEquipmentManager Version/commit

14b2b6a

### Node Version

v16.16.0

### Platform

Linux nixie-poolcontroller 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

### RS485 Adapter

BAS passthrough

### Are you using Docker?

- [ ] Yes.

### OCP

Nixie Standalone

### Pump(s)

Pentair Challenger 1 Speed

### Chlorinator(s)

AquaRite

### What steps will reproduce the bug?

Leave njsPC running for more that one day.

### What happens?

 The spa schedule (ID 1) in the attached replays will turn on between 0900-0930 as intended on the first day after starting njsPC.  The next day, the schedule does not turn on.

### What should have happened?

Schedule ID 1 (Spa) should turn on daily between 0900-0930.

### Additional information

I do not see any logs in the replay at 0900 the second day.  Restarting njsPC while the schedule should be running seems to address the issue, until the next day.  Trying to nail down the specifics has been difficult, since it takes a day to test.

Could this be related to https://github.com/tagyoureit/nodejs-poolController/discussions/963?  I will start with a fresh install if needed but would prefer to try to identify the issue first.

[2024-06-11_11-05-53.zip](https://github.com/user-attachments/files/15823410/2024-06-11_11-05-53.zip) (Spa schedule triggered correctly)
[2024-06-12_09-20-02.zip](https://github.com/user-attachments/files/15823411/2024-06-12_09-20-02.zip) (Spa schedule did not trigger)

rstrouse, 2024-06-19:
You have a pool and spa schedule that overlap.  Are you running the spa to just run water through the pipes.  If you are you should consider setting up a spillway circuit.  This way it is not actually engaging the spa.

celestinjr, 2024-06-19:
My intention was for the spa schedule to end when the pool schedule starts, not for them to overlap. 

I prefer to run the spa for a few minutes a day at least to circulate water through the bottom and spa suction piping.   Can those two schedules not share an “end (spa) / start (pool)”time?

I noticed the “triggered” property never resets to false for my spa schedule. Reading some other discussions, I believe it should be reset to false at the end of the schedule and/or at midnight. Is this correct?

rstrouse, 2024-06-20:
@celestinjr that is correct.  When it recalculates the next schedule date it should be setting the triggered flag to false.

celestinjr, 2024-06-20:
Based on poolState.json snapshots, it does not appear to be resetting the triggered flag for the spa circuit (circuit 1).  The below are 2 minutes apart just before and after midnight last night.

I'll capture a replay at that time if it would be helpful.

[image]

rstrouse, 2024-06-26:
You are correct on this.  I will get some time this weekend to take a look at it.  The new start/end times have been calculated but the triggered flag needs to be reset when there is a new start time.

celestinjr, 2024-07-30:
Looks like this is happening because the pool schedule switches on prior to the spa schedule end time and implicitly switches off the spa (without resetting the triggered flag).  The spa schedule end is never processed, because the spa was turned off already. 

Not sure yet the best method of fixing this, but I figured I’d share my findings. 

For now I just changed my config to end the spa schedule a minute earlier than the pool schedule starts.

TSquared58, 2024-08-02:
@rstrouse Is the @celestinjr 1 minute workaround above required because schedules run 1 minute longer than expected?

I have a Nixie Single Body running on a RPi 3B. I created a schedule to run my VSF pump from 9:00PM to 9:02PM (2 minutes). The poolState.json file startTime and endTime for this schedule are:

        "startTime": "2024-08-01T21:00:00.000-0700",
        "endTime": "2024-08-01T21:02:59.999-0700",
 
The log I captured shows the pump starting and ending:

       [01/08/2024, 21:00:02] info: NCP: Setting Pump IntelliFlo VSF to 1500 RPM.
       .
       .
       .
       [01/08/2024, 21:03:02] info: NCP: Setting Pump IntelliFlo VSF to 0 RPM.

This shows the pump running for 3 minutes when I would expect it to only run for 2 minutes.

Shouldn't the poolState endTime be 21:01:59.999?

rstrouse, 2024-08-02:
The end time is inclusive and the start time is not.  It is set up this way to be in parity with connected OCPs.  I think the issue is that the spa body schedule shuts down the pool schedule and does not restart it because the body circuits are mutually exclusive in a shared body.  It needs to untrigger the pool when it is interrupted by a spa schedule.  The act of the spa ending should switch to pool mode instead of off.