tagyoureit/nodejs-poolController #1027: [BUG] Cannot update PH or ORP Tank Levels

SNMKelley, 2024-09-18:
### nodejs-poolController Version/commit

8.0.4

### nodejs-poolController-dashPanel Version/commit

8.0.0

### Node Version

v18.17.0

### Platform

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Are you using Docker?

- [ ] Yes.

### OCP

EasyTouch 2 4P

### Chlorinator(s)

Intellichem

### What steps will reproduce the bug?

Navigate to Chemistry/Intellichem
Click on the Chlorine or Acid Tank and change the value to set the level
Click Save

### What happens?

TypeError: Cannot read properties of undefined (reading 'tolerance')
    at TouchChemControllerCommands.setIntelliChemAsync (/root/nodejs-poolController/controller/boards/EasyTouchBoard.ts:3044:48)
    at TouchChemControllerCommands.setChemControllerAsync (/root/nodejs-poolController/controller/boards/SystemBoard.ts:5049:49)
    at TouchChemControllerCommands.setChemControllerStateAsync (/root/nodejs-poolController/controller/boards/SystemBoard.ts:5065:46)
    at /root/nodejs-poolController/web/services/state/State.ts:86:61
    at Layer.handle [as handle_request] (/root/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at next (/root/nodejs-poolController/node_modules/express/lib/router/route.js:144:13)
    at Route.dispatch (/root/nodejs-poolController/node_modules/express/lib/router/route.js:114:3)
    at Layer.handle [as handle_request] (/root/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at /root/nodejs-poolController/node_modules/express/lib/router/index.js:284:15
    at Function.process_params (/root/nodejs-poolController/node_modules/express/lib/router/index.js:346:12)
    at next (/root/nodejs-poolController/node_modules/express/lib/router/index.js:280:10)
    at /root/nodejs-poolController/web/Server.ts:753:25
    at Layer.handle [as handle_request] (/root/nodejs-poolController/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/root/nodejs-poolController/node_modules/express/lib/router/index.js:328:13)
    at /root/nodejs-poolController/node_modules/express/lib/router/index.js:286:9
    at Function.process_params (/root/nodejs-poolController/node_modules/express/lib/router/index.js:346:12)

### What should have happened?

The value for the tank level should have changed to the desired value.

tagyoureit, 2024-09-23:
Can you attach a replay?  This should be getting set from the Intellichem.

SNMKelley, 2024-09-25:
[njsPC_error.zip](https://github.com/user-attachments/files/17138952/njsPC_error.zip)

SNMKelley, 2024-09-25:
I uploaded a zip of the replay.  Hope it is what you are looking for.

All the best
Sean

tagyoureit, 2025-07-09:
Download v8.1.1 and try again.

SNMKelley, 2025-07-09:
Sadly I still receive the same error after pulling both the controller and dashPanel.

SNMKelley, 2025-07-12:
[replay (3).zip](https://github.com/user-attachments/files/21197215/replay.3.zip)

SNMKelley, 2025-07-12:
> Can you please do another packet capture, but include all the downloaded
> files (esp poolstate.json and poolconfig.json). I need to see what your
> data looks like to figure out the exact issue.

I uploaded the complete capture zip to the issue.

tagyoureit, 2025-07-16:
Pull again.

SNMKelley, 2025-07-16:
Latest push fixed the issue!  Tusind Tak!

closed 2025-07-16 by SNMKelley