tagyoureit/nodejs-poolController #1040: Mode selection for UltraTemp

spifftek70, 2024-11-04:
## 💥 Proposal

### What feature you'd like to see

A selector in the API and UI for switching between the two modes of an UltraTemp depending on the body being used.

## Motivation

I have a shared body using the Nixie version of njsPC as I have a Pool and Spa that both use my UltraTemp to heat and cool them.

## Pitch

The UltraTemp from Pentair has two modes: Pool and Spa. Add a selector to the UI and API that changes the UltraTemp mode depending on the body in use.

tagyoureit, 2024-11-19:
Set the Ultratemp to `Pool/Spa`.
[image]

And then you set the setpoint for either body from the dashboard (or api):
[image]

[image]

When you are in pool mode it will use the heat settings from the pool circuit and the same for spa mode.

spifftek70, 2024-11-23:
My setting look a bit different than yours..  If this works, shouldn't the settings on the UltraTemp itself change to reflect that?  Because that isn't happening. ??

[image]

closed 2025-12-12 by tagyoureit