tagyoureit/nodejs-poolController #1044: [BUG] Using Screenlogic doesn't disable serial integration

trevorkwhite, 2024-11-17:
### nodejs-poolController Version/commit

8.0.4

### Are you using Docker?

- [ ] Yes.

### What steps will reproduce the bug?

Enable Screenlogic via the GUI or via config.json and run `npm start`

### What happens?

Appears the app continues to attempt a direct connection every 10 seconds, creating log spam and extra work:

`error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds`

### What should have happened?

Enabling Screenlogic should disable the local serial connection, preventing the log spam and retries.

### Additional information

If anyone wants to show me where to look, I'll cut a PR to fix this.

tagyoureit, 2024-11-19:
I'm not able to replicate this when setting the connection type to screenlogic.
[image]

trevorkwhite, 2024-11-19:
I used the same setup - let me know if you want a copy of the config json file - here's what I see in the logs.

```
[11/16/2024, 5:10:03 PM] info: The current git branch output is master
[11/16/2024, 5:10:03 PM] info: The current git commit output is 52942cd8b1c7c489cd9f0dbfad0c142d0cd735be
[11/16/2024, 5:10:03 PM] info: Starting up SSDP server
[11/16/2024, 5:10:03 PM] info: Server is now listening on 0.0.0.0:4200 - 192.168.2.173:4200
[11/16/2024, 5:10:03 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:10:03 PM] info: Checking njsPC versions...
[11/16/2024, 5:10:03 PM] info: Starting Pool System easytouch
[11/16/2024, 5:10:03 PM] info: Auto-backup initialized Last Backup: 1969-12-31T16:00:00.000-0800
[11/16/2024, 5:10:04 PM] info: Screenlogic: Unit Pentair: 20-1A-BB found at 148.62.14.29:8152
[11/16/2024, 5:10:04 PM] info: Screenlogic: connect to Pentair: 20-1A-BB POOL: 5.2 Build 738.0 Rel at 148.62.14.29:8152
unused valve, loadCenterIndex = 0 valveIndex = 0
Possible unknown string: �l
[11/16/2024, 5:10:05 PM] warn: Comms port 0 is not open. Message aborted: 165,33,16,33,209,1,1,1,202
[11/16/2024, 5:10:05 PM] warn: Message aborted after 0 attempt(s): 165,33,16,33,209,1,1,1,202
[11/16/2024, 5:10:05 PM] info: Cannot set/delete eggtimer on circuit 8.  Error: Comms port 0 is not open. Message aborted: 165,33,16,33,209,1,1,1,202
[11/16/2024, 5:10:05 PM] warn: dt:Sat Nov 16 2024 17:18:07 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[11/16/2024, 5:10:05 PM] warn: dt:Sat Nov 16 2024 17:18:07 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[11/16/2024, 5:10:05 PM] warn: dt:Sat Nov 16 2024 17:18:07 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[11/16/2024, 5:10:05 PM] warn: dt:Sat Nov 16 2024 17:18:07 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[11/16/2024, 5:10:05 PM] warn: dt:Sat Nov 16 2024 17:18:07 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
[11/16/2024, 5:10:05 PM] warn: dt:Sat Nov 16 2024 17:18:07 GMT-0800 (Pacific Standard Time) lat:undefined lon:undefined Not enough information to calculate Heliotrope.  See https://github.com/tagyoureit/nodejs-poolController/issues/245
Possible unknown string: �l
[11/16/2024, 5:10:13 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:10:13 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:10:23 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:10:23 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:10:33 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:10:33 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:10:43 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:10:43 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:10:53 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:10:53 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:11:03 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:11:03 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:11:13 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:11:13 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:11:23 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:11:23 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:11:33 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:11:33 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:11:43 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:11:43 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
[11/16/2024, 5:11:53 PM] warn: Inactivity timeout for 0 serial port /dev/ttyUSB0 after 10 seconds
[11/16/2024, 5:11:53 PM] error: Error opening port 0: Error: No such file or directory, cannot open /dev/ttyUSB0. Retry in 10 seconds
```

Those warnings / errors continue every 10 seconds, despite Screenlogic being the primary port

commit a9f68016 (2024-11-23): Fix #1044 fix for screenlogic port; fix schedule logic
closed 2024-11-23 by tagyoureit