tagyoureit/nodejs-poolController #1046: Select What is Shared in Nixie Shared Body

spifftek70, 2024-11-23:
## 💥 Proposal

### What feature you'd like to see

Allow users to select what features/circuits they share in "Shared Bodies"

## Motivation

I have a pool and spa that share an UltraTemp heater via actuator valves, but each have their own filters. The pool is a DE filter and the spa is a cartridge filter.  I want the ability to monitor the pressure in BOTH filters, but currently, Nixie Shared body does not allow for that.  It only allows for ONE filter in a Shared Body. 
Some pools and spas only share the UltraTemp, while each have their own filter.  Some share filters, but NO heater.  Others share both.

## Pitch
Add the ability for users to select those features/circuits they share in Nixie Shared Body like so: 
[  ] Filter
[X] Heater

tagyoureit, 2024-11-26:
Do you have a spillover from the spa to the pool?

This isn't a very common setup.  You can do pretty much everything today including adding multiple heaters, but you can't add a 2nd filter to a shared body.  It would take quite a bit of work to do that.  If all you want to do is monitor the 2nd filter psi you would be best going with HomeAssistant or another iot service.  REM can send the PSI directly there.

spifftek70, 2024-11-28:
No spillover at all.. there is a water refill line from the pool to the spa, but that is it.

Okay, the both filter gauges are currently wired into the 8-layer stackable HAT.  How do I set that up with IoT or HomeAssistant?

tagyoureit, 2024-11-28:
REM can output a feed in http or mqtt messages.  These can be fed directly into home assistant or other home automation software.  The specifics of that are beyond the scope this project.

spifftek70, 2025-02-02:
Question:  I could use `Nixie Duel Body`, but for some reason, `Pump Off During Valve Action` doesn't seem to work when I select the `Duel Body` option.  Any ideas?

tagyoureit, 2025-02-02:
We only delay pumps when the pool or spa circuit(/valve) is changed and the pool valve is diverted or the spa valve is not diverted.