tagyoureit/nodejs-poolController #1047: [BUG] Message aborted - unable to connect to

christensonautohome, 2024-11-26:
### nodejs-poolController Version/commit

8.0.5

### nodejs-poolController-dashPanel Version/commit

8.0.0

### Node Version

v18.20.4

### Platform

Pi 3b+

### RS485 Adapter

GearMo Mini USB to RS485

### Are you using Docker?

- [ ] Yes.

### Pump(s)

EasyTouch2 8

### Chlorinator(s)

iChlor IC30

### What steps will reproduce the bug?

I'm unable to make any changes though the Pool Controller Dashboard but the dashboard does receive updates and shows correct data from the outside pool controller.

### What happens?

If i select a feature or turn on/off the pool the button will turn yellow then I receive: Error:OutboundMessageError
MessageMessage aborted after 4 attempt(s): 165,7,16,33,134,2,6,1,1,108
ApiError: Message aborted after 4 attempt(s): 165,7,16,33,134,2,6,1,1,108 
    at RS485Port.writeMessage (/home/pi/nodejs-poolController/controller/comms/Comms.ts:1013:31)
    at RS485Port.processWaitPacket (/home/pi/nodejs-poolController/controller/comms/Comms.ts:949:22)
    at RS485Port.processOutboundPackets (/home/pi/nodejs-poolController/controller/comms/Comms.ts:957:19)
    at RS485Port.processPackets (/home/pi/nodejs-poolController/controller/comms/Comms.ts:939:14)
    at Timeout. (/home/pi/nodejs-poolController/controller/comms/Comms.ts:992:52)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)

### What should have happened?

the button on the controller dashboard should have turned green if turning on or no color if turning off.  However I get yellow then the error pops and the pool doesn't respond.

### Additional information

I ran a test on the RS-485 adapter and receive the expected ffa5ff repetition so that seems to be working and I also receive information from the pool, just seems like commands are not going out to the pool.

christensonautohome, 2024-11-26:
I've already ordered a new RS-485 adapter thinking that might be the issue but wasn't sure if this is an expected error encountered when a RS-485 adapter is failing.

tagyoureit, 2024-11-26:
Did this just start with 8.0.5?  Or does it also happen with the previous release?  Nothing in this latest patch should affect that.

christensonautohome, 2024-11-27:
No, I don't often control my pool, so I'm not sure when it might've started. I just recently updated to the latest release to see if that might fix the issue.

christensonautohome, 2024-11-28:
I think it's boiling down to an issue with my R Pi 3B+; going to take some time two swap out for a new one and update when I've figured out the issue.

christensonautohome, 2024-12-01:
swapped out RS485 to USB stick and the Pi hosting everything.  No clue what it was at the end of the day but a full resupply of parts and fresh install on new SD card fixed it right up.  closing ticket

closed 2024-12-01 by christensonautohome