tagyoureit/nodejs-poolController #1060: Add support for VGreen/Century drop-in replacement VSP motors

Ixian, 2025-02-10:
## 💥 Proposal

### What feature you'd like to see

Drop-in VSP motor replacements by Century (Regal Beloit) that all use the same controller - supports modbus/rs-485 and can speak the Jandy protocol. Here's a link to the one I have: https://a.co/d/euEak43

There's actually a ton of info related to this motor available, and someone already did an ESPHome integration with it: https://github.com/gazoodle/CenturyVSPump

I could just use that, but I'd really rather have it all running off my poolController PI vs. adding another random ESP32 to the mix that is control of the most important part of my pool (the filter pump). 

And the communications protocol is completely documented in this pdf: https://www.troublefreepool.com/attachments/gen3-epc-modbus-communication-protocol-_rev4-17-pdf.358807/

More information and background here:  https://www.troublefreepool.com/threads/century-regal-vgreen-motor-automation.238733/

Would it be possible to add to Nixie(?) 

## Motivation

I imagine the crossover between this project and the kind of people who buy and install these is large - most owners like myself in this situation don't have automation either and are installing a VSP as step one. 

## Pitch

I'm willing to dive in and help, work on a PR, if someone wants to give me some pointers on where it needs to be properly referenced etc. so I don't have to puzzle out how to add a new pump to this codebase by myself :)

tagyoureit, 2025-02-14:
There have been a couple of threads about this (and similar) pumps.  https://github.com/tagyoureit/nodejs-poolController/discussions/686 and https://github.com/tagyoureit/nodejs-poolController/discussions/523.

Users here didn't have success with their pumps, but the TFP thread you referenced does have instructions for changing the protocol to Jandy.  Maybe a similar approach would allow the Pentair protocol to 'stick'?  Then you wouldn't need any code changes.

Adding an entirely new protocol (modbus) is a big undertaking.   Even if you are an expert programmer, adding that support here would be hard.

mguinness, 2026-03-02:
@celestinjr I know you have the V-Green EVO, do you think your changes would also work with the Nidec Neptune NPTT270?

[Neptune RS-485 Connection Guidelines](https://images.inyopools.com/cloud/documents/NEPTUNE-RS485-CONNECTION-GUIDELINES.pdf)

[Neptune Modbus Register Specification](https://images.inyopools.com/cloud/documents/NPTQ270-MODBUS-REGISTER-SPECIFICATION-VS-MOTOR.pdf)

whizkidTRW, 2026-04-06:
I can confirm that the Century Modbus protocol works flawlessly! I was able to stack my Aquarite SWG Chlorinator and the Century VGreen pump on to the same RS485 bus and control them simultaneously with njs-PC. So great to have it all in one place now, instead of the pump on a dedicated esp32 module. Thank you so much!

closed 2026-10-07 by tagyoureit