tagyoureit/nodejs-poolController #1076: [BUG] Pool Circuit not stopping at sunset

ddyer8, 2025-04-24:
### nodejs-poolController Version/commit

v.8.0.4

### nodejs-poolController-dashPanel Version/commit

v.7.6.1

### relayEquipmentManager Version/commit

v.8.1.0

### Node Version

v18.15.0

### Platform

6.1.21-b8+ #1642, aarch64 GNU/Linux

### RS485 Adapter

DTECH USB to RS485

### Are you using Docker?

- [ ] Yes.

### OCP

None

### Pump(s)

Superflo VST (configured as Intelliflo VS)

### Chlorinator(s)

Aquarite

### What steps will reproduce the bug?

Use dashpanel to create a schedule for the Pool circuit, which ends at sunset

### What happens?

When ending at sunset, the pool continues to run.  If I change the setting to manual and select a time, it cuts off appropriately.  

### What should have happened?

Pool should turn off at sunset.

### Additional information

I have another circuit (lights), which come on at sunset, and that works appropriately.

ddyer8, 2025-04-24:
I think this was also an issue with dashpanel v.8.0.0, but I'll upgrade and try again.  I reverted to v.7.6.1 in an attempt to get the hardware tab back on REM Chem (https://github.com/rstrouse/nodejs-poolController-dashPanel/issues/89)

ddyer8, 2025-04-28:
Still an issue with v8.0.0

tagyoureit, 2025-05-02:
Have you tried the latest - v8.1.0?

ddyer8, 2025-05-03:
REM? I’m running 8.1. I don’t see that tag for the other apps. I’ve been having the issue for a few versions. Any troubleshooting ideas?Thank you,Dustin

tagyoureit, 2025-05-09:
The latest version of the main app is v8.1.0.  https://github.com/tagyoureit/nodejs-poolController/blob/master/package.json
DashPanel is 8.1.  REM is still on v8.0.

ddyer8, 2025-05-09:
Seeing the other way (v8.0 dash and 8.1 REM), but I’m on latest versions. Is there a workaround to set the tank level manually without the browser popover?Any other info I can provide to help troubleshoot? Appreciate the app and support. Thank you,Dustin