tagyoureit/nodejs-poolController #1114: RS485 Not initializing via Mega-Bas

Exit2Studios, 2025-07-17:
### nodejs-poolController Version/commit

7.3.1-rc1

### nodejs-poolController-dashPanel Version/commit

7.3.1-rc1

### relayEquipmentManager Version/commit

Latest from master branch as of July 2025

### Node Version

v18.20.2

### Platform

Linux poolai 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr 3 17:20:52 BST 2023 aarch64 GNU/Linux

### RS485 Adapter

Sequent Mega-BAS (via REM)

### Are you using Docker?

- [ ] Yes.

### OCP

None / Nixie (standalone setup)

### Pump(s)

Intelliflo3 (using VSF in drop down)

### Chlorinator(s)

None

### What steps will reproduce the bug?

	1.	Set COM port in DashPanel to /dev/ttyS0 (RS485 passthrough via REM + Mega-BAS)
	2.	Restart nodejs-poolController
	3.	Observe failure due to /dev/ttyS0 not being present (REM does not expose the port as a valid tty device)
	5.	poolController throws: ApiError: Unable to close the current RS485 port

### What happens?

	•	Once a COM port is selected that does not exist (/dev/ttyS0), poolController cannot release it.
	•	Subsequent attempts to change the COM port via GUI are blocked by ApiError: Unable to close the current RS485 port, even if the new port is valid.
	•	This leaves the system in a non-functional state until config files are manually edited and/or poolController is restarted after stopping REM

### What should have happened?

	•	Once a COM port is selected that does not exist (/dev/ttyS0), poolController cannot release it.
	•	Subsequent attempts to change the COM port via GUI are blocked by ApiError: Unable to close the current RS485 port, even if the new port is valid.
	•	This leaves the system in a non-functional state until config files are manually edited and/or poolController is restarted after stopping REM

### Additional information

This was discovered while attempting to use REM’s Mega-BAS card for RS485 passthrough. Although REM detects the Mega-BAS correctly, it does not expose /dev/ttyS0 to the OS, which causes poolController to fail. 

Let me know if logs or packet captures are needed — I can reproduce the state consistently.

tagyoureit, 2025-07-22:
> nodejs-poolController Version/commit
7.3.1-rc1

Are you really on this version?  Current is 8.1.2 and this is a few years old.  And a replay would be needed for any debugging.

Exit2Studios, 2025-07-22:
Replay Attached:
[replay.zip](https://github.com/user-attachments/files/21371619/replay.zip)

Exit2Studios, 2025-07-23:
I've done a fresh install, with only the Pi5 and the Mega-BAS installed - connected to pump via a brand new USB to RS485 adapter.

Here is the replay:

[replay (1).zip](https://github.com/user-attachments/files/21393866/replay.1.zip)

tagyoureit, 2025-07-24:
I would start by turning off MQTT.

This is trying to set the pump speed through an Aqualink controller.  Do you have an Aqualink controller?
```
[23/07/2025, 14:05:06] silly: MQTT send:
topic: aqualinkd/Filter_Pump/set
message: 0
opts:{"retain":true,"qos":0}
```

I also see 0 (zero) inbound packets.

And there are calls only milliseconds apart to turn the pool on/off.  Where's that coming from?
```
{"dir":"in","proto":"api","requestor":"192.168.1.17","method":"PUT","path":"/state/circuit/setState","body":{"id":6,"state":false},"ts":"2025-07-23T14:05:41.315-0500"}
...
{"dir":"in","proto":"api","requestor":"192.168.1.17","method":"PUT","path":"/state/circuit/setState","body":{"id":6,"state":true},"ts":"2025-07-23T14:05:43.153-0500"}
...
{"dir":"in","proto":"api","requestor":"192.168.1.17","method":"PUT","path":"/state/circuit/setState","body":{"id":6,"state":false},"ts":"2025-07-23T14:05:45.012-0500"}
{"dir":"in","proto":"api","requestor":"192.168.1.17","method":"PUT","path":"/state/circuit/setState","body":{"id":6,"state":true},"ts":"2025-07-23T14:05:45.885-0500"}
```

When you start the packet capture, you probably don't want to reload the entire configuration.  That will wipe all your settings.
```
[23/07/2025, 14:05:06] info: Resetting System to initial defaults
```

Exit2Studios, 2025-07-24:
@tagyoureit, the repeating put requests are coming from my laptop browser, so I could only assume that is coming from DashPanel or REM?

I will disable mqtt (and no , I don't have an Aqualink system and I have no idea how that got enabled) - and get you a clean capture.

Thanks as always for the amazing support!

Exit2Studios, 2025-07-24:
I went ahead and did another capture while I messed with the physical pump.  I turned pump on/off (from breaker), did the "reset" (hold quick clean, on/off, connect), pump on/off again, paired pump back to app.

Interesting that there are a few bites received by the rs485.  Progress??

[replay (3).zip](https://github.com/user-attachments/files/21415222/replay.3.zip)

tagyoureit, 2025-07-24:
Unfortunately, there is nothing new here.  4 bytes may just be the port initializing, or a little power blip on the line.  It's not enough to have a full packet.  

You can set the activity timeout much, much higher.  That would at least avoid the port opening and closing so often.  But that has nothing to do with your pump not responding.

You need to focus on three separate areas, as these are where things might go wrong:
1. The rs485 port isn't sending out (or receiving) data properly.  Maybe because of a hardware or a software error.
2. The pump isn't receiving the packets.  Possibly due to a bad connection or break in the wire.
3. The pump isn't listening for the packets or not receiving them.  This could be any number of configuration settings on the pump itself.

If I were you, my next step would be to dig into number 1.  To do this, I would start njsPC and have it control one port (eg /dev/ttyUSB0).  Setup another computer/terminal/whatever to be listening on your other board (/dev/ttyS0).  Connect them together.  See [wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details) for some ideas here.  If this works, then you have eliminated 1 and 2 from the the list and you can focus on the pump itself.  If it doesn't work, try reversing the ports, swapping wires, replacing wires, etc.